jbaumann/attiny_daemon #1: Multiple-Cell Batteries...

starhawk64, 2019-08-31:
Apologies for contacting you here, but I can find NO other way to get ahold of you (please create or otherwise expose a public-facing email address! It would be very helpful.).

I want to implement this device, using a USB power bank "kit" (a plastic housing and the power bank controller PCB, with room for up to six 18650 cells) and some 18650 cells from old laptop batteries. HOWEVER, I can't tell how to interface / monitor multiple batteries at once, and I'd be concerned about the batteries' respective charges getting out of whack -- I don't know much about LiPoly cells, but I know they have a nasty temper and I know that balancing cells is a concern. (...and, yes, I also know better than to solder to batteries. While I've not read up on the latest LiPoly concerns, I'm mildly experienced as an electronics hobbyist.)

Example listing for power bank kit (mine is identical, but not from that listing) --> https://www.ebay.com/itm/183897827870

jbaumann, 2019-09-03:
Sorry for not exposing the e-mail address, should work now.

To me it seems that the kit is doing all the balancing so that should be taken care of, as far as I can see. However, you would need to connect the Vcc of the ATTiny_Daemon to the balanced output of the batteries inside the power bank. Do you have a schematic of the power bank you want to use?

starhawk64, 2019-09-03:
I don;t need schematics in this case -- it's a "bring your own batteries"
kit. I thought I was more clear about that, sorry...

jbaumann, 2019-09-03:
Ok, let us assume that the batteries are balanced by your power bank, for charging and discharging. Then each one of the batteries should have roughly the same charge and thus roughly the same voltage. So simply solder one wire to the positive terminal of the battery holder for one of the batteries and then use that as Vcc for the ATTiny_Daemon (assuming that GND is connected to the negative terminal). Since the current needed by the ATTiny_Daemon is very small compared to the needs of the RPi this shouldn't unbalance the batteries unduly.

starhawk64, 2019-09-03:
OK, great! Thanks :) I'm really looking forward to making this...

One other question... how long can the I2C wires be? What I'm building
requires a short distance there. I was thinking of using optocouplers to
make an I2C-to-current loop adapter (since current loops can go for truly
insane lengths of distance when they have to) but I don't; want to rig that
if I don't have to. I think I can keep it to less than six or eight inches.

jbaumann, 2019-09-03:
I'm using normal Dupont wires with 20cm length quite often when experimenting with I2C. This should be around 8in. Since I used a CRC to check the correctness of the transferred data and a simple retransmit in case of errors this should work with even longer cables without problems (I think that 1m is given as a maximum quite often...).

starhawk64, 2019-09-03:
Okay, awesome! Thanks for the help :)

jbaumann, 2019-09-05:
Only to be sure: Check that GND and the negative terminal of the battery are connected. Otherwise it is not that simple... And let me know how it goes.

closed 2020-03-23 by jbaumann