jbaumann/attiny_daemon #2: Connections

st599, 2019-09-17:
Hi

I can't see from your documentation where the wires from the modified board are connected to or where exactly the Dupont connector attaches.

Could you provide images of the final layout?

jbaumann, 2019-09-20:
Thank you for the feedback, I will write an additional Wiki entry as soon as I have time.

If you look at the layout from the bottom side, I have labelled the different connections.  Here is a link to the picture: https://github.com/jbaumann/attiny_daemon/blob/master/hardware/v1.8shim/Photo_Bottom_View.png

If you look at this description of the board modification: https://github.com/jbaumann/attiny_daemon/wiki/Modifying-the-Geekworm-UPS-HAT

you connect the red cable (soldered to the battery power) to _Vcc_ (the connection on the right in the picture of the board) and the orange cable (soldered to the switch connection) to the pin labeled _Sw_ (the third from the right on the picture.

If you want to connect a button you simple connect it to Btn and GND, and if you want to measure an additional external voltage you connect it to Ve.

Does this help until I write the documentation?

adrianmihalko, 2019-12-16:
@jbaumann thanks for this great project. I aggree that the Wiki is covering many thing, but it's a little bit unclear to me. There should be a single page with 

- modification of the Pi, needed to install the PCB
- complete assembly of the PCB and it's installation on the Pi
- Software installation

Anyone has one or two PCB free available in the EU? If I order now, I need to wait 1 month from china. :(

jbaumann, 2019-12-18:
Adrian, thank you for your praise and your suggestion. I‘m not sure whether a single page with all the information is better than multiple pages, but I can try and write a step-by-step guide that references the other pages that I have already written.

The Pi does not need any modification, which was in fact one of my goals when creating the project. The UPS has to be modified though.
The assembly of the PCB is extremely simple and the silk screen printed on the PCB even gives the values for every component, so this shouldn‘t be a problem, but I‘ll look into writing something up with a few pictures (e.g., components with a low height first, how to bend the connector and the LED etc.). 
One question, though, without being disrespectful: If someone is able to follow the instructions to modify the UPS, shouldn‘t they also be able to assemble the board?

The software installation is already described in detail in this page: https://github.com/jbaumann/attiny_daemon/wiki/Installation-on-the-Raspberry-Pi

My experience with JLCPCB (I‘m not affiliated with them in any way) is very good and so far I have received the PCBs I ordered in 2 weeks (more or less). So it is not as bad as you think, but you won‘t be receiving the PCBs before Christmas.

jbaumann, 2020-01-06:
I have now described the assembly of the board in details, please have a look: https://github.com/jbaumann/attiny_daemon/wiki/Assembling-the-ATTiny_Daemon-Hardware

jbaumann, 2020-01-06:
I have described the connections to the board in detail on the following page: https://github.com/jbaumann/attiny_daemon/wiki/Connecting-the-ATTiny_Daemon-Hardware

Please have a look

adrianmihalko, 2020-01-08:
Wow, it's way more clear now. Thanks for the update, parts ordered.

closed 2020-03-23 by jbaumann