jbaumann/attiny_daemon #3: Support for Geekworm's new UPS HAT2?

poky, 2019-10-12:
It appears Geekworm has released their RPi4B supported UPS HAT

https://geekworm.com/products/raspberry-pi-ups-hat-2

It also mentioned the I2C power off function, does anyone on here have tried?

jbaumann, 2019-10-12:
It sounds pretty interesting, and if it works as promised then we might be able to use it with ATTiny_Daemon directly, without the hardware modification.

I haven’t read about an I2C power-off feature, only the auto-power-on. Was it this feature you meant?

I have contacted Geekworm and asked for further information (chips, schematics, etc.). If it sounds ok I will order (at least) one to test it...

jbaumann, 2019-10-12:
And already got an answer. They “cannot” provide any information. I looked at the photo of the UPS2, and I cannot read anything on the chips, but they also have a Github repo for it https://github.com/geekworm-com/UPS2. They use a power-off dtoverlay.

jbaumann, 2019-10-12:
I have been offered a sample to experiment with, looking forward to test it.

akula3, 2019-12-03:
Hello Joachim!
I'm looking for an UPS with the same features like you.
Your project is very interesting, but I wouldn't trust me to solder on SMD's.
Did you make already some positive experience with the UPS HAT2 from geekworm and can share that?
Danke schön :-)

jbaumann, 2019-12-11:
I have now done the first tests and the preliminary conclusion is that it might be usable as an UPS. It will need the ATTinyDaemon to work properly though.
Caveats: The charge amperage should be changed from 2A (default setting) to 1A. To do this a solder bridge has to be removed and another one soldered on two pads. This shouldn't be a problem even for beginners.
I would also enable the additional signalling that the RPi has shut down, and this needs another solder bridge.
Detailed information will follow in a few weeks...

And akula3, you should never try to charge a LiPo battery using a Li-Ion charging circuit.

jbaumann, 2020-03-29:
I just created a new release that supports the Geekworm UPS2. I currently use it successfully and can recommend it. One modification is necessary though. You should change the maximum charge current to 1A, otherwise the output voltage will dip for a short time sometimes leading to crashes. And you really need to use a battery with at least 2.500mAh (to provide the maximum needed amperag).
I will try and find the time to write up a review and how to install everything (much simpler than with the Geekworm UPS 1, you only need to solder a cable to the switch itself and one to the battery pad with the positive voltage).

jbaumann, 2020-04-11:
It took some time to write it up but you can find the short review and the needed modifications in the Wiki.

closed 2020-04-11 by jbaumann