jbaumann/attiny_daemon #4: Help needed

rahmanshaber, 2020-02-06:
Hi, I am developing a portable pi UMPC/sidekick. I want to add your solution there and forget about developing one from ground. 
So i need Help of yours to get things right.
1. i will use a existing battery charging breakout board to charge the battery. [this one](https://www.aliexpress.com/item/32798858483.html)
2. so can i use the the ups part form your project to power off the system from raspberry pi and a button to boot the pi?
3. i want to see the battery details in with regular way like you can see the laptop battery details in app like xfce-battery or lxqt-battery plugin for panel . Or a terminal way is ok.

Please help me out
Here is my project.
https://mutantc.gitlab.io/

jbaumann, 2020-02-14:
Hi,

I‘m sorry for the delay in answering your e-Mail. I‘m currently on vacation and will look into it after I‘m back. Without looking into the details most of what you need should be possible. Starting the Pi using abutting might be a bit more complicated but let me think about that when I‘m back.

Cheers, Joachim

rahmanshaber, 2020-02-16:
thanks. I am waiting.

jbaumann, 2020-03-23:
The button should work as you need it (it was already implemented), and the data can be read using the Python class Attiny_i2c. Its use is shown in the script useATTiny.py. You can use this to display the needed information however you want it.

closed 2020-03-23 by jbaumann