jbaumann/attiny_daemon #5: Couldn't read 16 bit register 0x1 correctly.

BaiGeorgi, 2020-02-15:
When i run "sudo systemctl status attiny_daemon" i get this error:

> pi@pi1:~ $ sudo systemctl status attiny_daemon
> ● attiny_daemon.service - ATTiny UPS Daemon
>    Loaded: loaded (/etc/systemd/system/attiny_daemon.service; enabled; vendor preset: enabled)
>    Active: active (running) since Sat 2020-02-15 15:56:40 GMT; 7s ago
>  Main PID: 631 (attiny_daemon.p)
>    CGroup: /system.slice/attiny_daemon.service
>            └─631 /usr/bin/python3 -u /opt/attiny_daemon/attiny_daemon.py
> 
> Feb 15 15:56:42 pi1 attiny_daemon.py[631]: ATTiny Daemon version 2.1.1
> Feb 15 15:56:43 pi1 attiny_daemon.py[631]: Couldn't read 16 bit register 0x1 correctly.
> Feb 15 15:56:43 pi1 attiny_daemon.py[631]: Couldn't read 16 bit register 0x1 correctly.
> Feb 15 15:56:44 pi1 attiny_daemon.py[631]: Couldn't read 16 bit register 0x1 correctly.
> Feb 15 15:56:44 pi1 attiny_daemon.py[631]: Couldn't read 16 bit register 0x1 correctly.
> Feb 15 15:56:45 pi1 attiny_daemon.py[631]: Couldn't read 16 bit register 0x1 correctly.
> Feb 15 15:56:45 pi1 attiny_daemon.py[631]: Couldn't read 16 bit register 0x1 correctly.
> Feb 15 15:56:46 pi1 attiny_daemon.py[631]: Couldn't read 16 bit register 0x1 correctly.
> Feb 15 15:56:46 pi1 attiny_daemon.py[631]: Couldn't read 16 bit register 0x1 correctly.
> Feb 15 15:56:47 pi1 attiny_daemon.py[631]: Couldn't read 16 bit register 0x1 correctly.

Can you tell me what is wrong?

jbaumann, 2020-02-15:
I‘m currently on vacation, but this means that not all read requests have been successful (which is quite normal and no problem) and that you have the debug flag still enabled in your configuration. As long as you only get the debug message and no warning everything is ok.
If you do not want to see these messages simply turn debug messages off as soon as you have verified that everything works ok.

BaiGeorgi, 2020-02-15:
Thank you for the information.

closed 2020-02-15 by BaiGeorgi