jbaumann/attiny_daemon #7: USIWire issue?

pawsnizzy, 2020-02-24:
Hi,
I am intent on implementing this device on my RPi, but am getting the following errors in the Arduino IDE:

`In file included from C:\Users\Dave\Documents\Arduino\libraries\USIWire\src\USI_TWI_Master/USI_TWI_Master.h:76:0,

                 from C:\Users\Dave\Documents\Arduino\libraries\USIWire\src\USIWire.cpp:27:

C:\Users\Dave\Documents\Arduino\libraries\USIWire\src\USI_TWI_Master/usi_io.h:1:1: error: expected unqualified-id before '.' token

 ../USI_TWI_Slave/usi_io.h

 ^

C:\Users\Dave\Documents\Arduino\libraries\USIWire\src\USIWire.cpp: In member function 'void USIWire::begin()':

C:\Users\Dave\Documents\Arduino\libraries\USIWire\src\USIWire.cpp:55:3: error: 'USI_TWI_Master_Initialise' was not declared in this scope

   USI_TWI_Master_Initialise();

   ^~~~~~~~~~~~~~~~~~~~~~~~~

C:\Users\Dave\Documents\Arduino\libraries\USIWire\src\USIWire.cpp:55:3: note: suggested alternative: 'USI_TWI_Slave_Initialise'

   USI_TWI_Master_Initialise();

   ^~~~~~~~~~~~~~~~~~~~~~~~~

   USI_TWI_Slave_Initialise

Multiple libraries were found for "USIWire.h"
 Used: C:\Users\Dave\Documents\Arduino\libraries\USIWire
Multiple libraries were found for "EEPROM.h"
 Used: C:\Program
exit status 1
Error compiling for board ATtiny25/45/85.
`
Can you suggest where I went wrong - these error messages might as well be in greek:(

Thanks

jbaumann, 2020-02-25:
Hi,

To me this looks like multiple libraries/library versions for the USIWire implementation in your library path. Could you check and verify that you have only one library version (and preferably the one that I used i.e., 0.1.0)?

The same seems to have happened with your eeprom library...

Cheers, Joachim

pawsnizzy, 2020-03-15:
No multople libaries at all - I've even gone so far as to delete EVERY library except the 2 above - and it's still the same

adrianmihalko, 2020-03-21:
I have same problem. Arduino 1.8.12. Without Usiwire library I get error about missing library (as expected)

```
In file included from /Users/adrianmihalko/Downloads/attiny_daemon-master/firmware/ATTinyDaemon/ATTinyDaemon.ino:1:0:
ATTinyDaemon.h:1:10: fatal error: USIWire.h: No such file or directory
 #include <USIWire.h>
          ^~~~~~~~~~~
compilation terminated.
exit status 1
USIWire.h: No such file or directory
```

Now, I install library as Sketch - Include Library - Add .ZIP Library and I choose USIWire-0.1.0.zip.

I click Very/Compile and I get the same error:

```
In file included from /Users/adrianmihalko/Documents/Arduino/libraries/USIWire-0.1.0/src/USI_TWI_Master/USI_TWI_Master.h:76:0,
                 from /Users/adrianmihalko/Documents/Arduino/libraries/USIWire-0.1.0/src/USIWire.cpp:27:
/Users/adrianmihalko/Documents/Arduino/libraries/USIWire-0.1.0/src/USI_TWI_Master/usi_io.h:1:1: error: expected unqualified-id before '.' token
 ../USI_TWI_Slave/usi_io.h
 ^
/Users/adrianmihalko/Documents/Arduino/libraries/USIWire-0.1.0/src/USIWire.cpp: In member function 'void USIWire::begin()':
/Users/adrianmihalko/Documents/Arduino/libraries/USIWire-0.1.0/src/USIWire.cpp:55:3: error: 'USI_TWI_Master_Initialise' was not declared in this scope
   USI_TWI_Master_Initialise();
   ^~~~~~~~~~~~~~~~~~~~~~~~~
/Users/adrianmihalko/Documents/Arduino/libraries/USIWire-0.1.0/src/USIWire.cpp:55:3: note: suggested alternative: 'USI_TWI_Slave_Initialise'
   USI_TWI_Master_Initialise();
   ^~~~~~~~~~~~~~~~~~~~~~~~~
   USI_TWI_Slave_Initialise
exit status 1
Error compiling for board ATtiny25/45/85.
```

jbaumann, 2020-03-23:
Ok, I'm not sure how this happens, but if you use "Sketch - Include Library - Add .ZIP Library", then the file contains only the following: "../USI_TWI_Slave/usi_io.h". This is clearly wrong and two possible explanations are on the top of my mind:
- The original author uses a symbolic link that is interpreted wrongly by the Arduino IDE when unpacking the files
- The Zip-Archive knows that both files are the same and saves only a reference, which again is interpreted wrongly by the Arduino IDE
In both case I would see the problem in the Arduino IDE.

Two simple possible fixes:
- Copy the file from the TWI_Slave over to the TWI_Master
- Remove the directory, unpack the library manually and copy it to your libraries folder

I suggest the second way as the general way to install libraries. This ensures that everything that goes wrong comes from your own errors.

puuu, 2020-03-24:
Hello, I am the original author of `USIWire`. Please note that I have released version 0.2.0, which no longer contains symbolic links. It is also back in the Arduino Library Manager.

PS: @jbaumann very amazing project! I had something like this in mind when developing USIWire. I will give it a try!

adrianmihalko, 2020-03-24:
Thank you guys, release 0.2.0 solved the issue.

closed 2020-03-24 by jbaumann