jbaumann/attiny_daemon #8: missing Wire.onInterrupt(disable_watchdog) routine, compilation terminated

deHarro, 2020-08-19:
Hi Joachim,

thanks for the elaborated design and DIY explanations for the RasPi UPS HAT!

Sadly I cannot get the firmware for the AtTiny to compile since there is missing code as stated above (Wire.onInterrupt(disable_watchdog)).
How to fix or circumvent this?
Thanks!

Harald

deHarro, 2020-08-19:
Hi Joachim,

I headed over to puuuu and added your [proposed code fragments](https://github.com/puuu/USIWire/issues/13) into my local copy of his USIWire implementation.

For now this resolves my issue, but how is your preferred solution to that?
Thanks!

Harald

jbaumann, 2020-08-19:
Hi Harald,

I have a fork in my repository that adds exactly this needed functionality. See https://github.com/jbaumann/USIWire.

Until my proposal for an extension is accepted the simplest way is to download this repository and unzip it in the library folder.

Cheers, Joachim

deHarro, 2020-08-19:
Hi Joachim,
ok, the usual missing "über den Tellerrand schauen" :)
I concentrated on the UPS stuff and missed the lib, sorry.

I try to implement your daemon with a Digispark module, but I run into issues when compiling for the Digispark:

- error: 'State' is not a class or namespace
- error: 'Register' is not a class or namespace
- error: 'Reset_Configuration::Value' is not a class or namespace

If I choose AtTiny85 as device, all is well.

Any advice on that?
Thanks!

Harald

jbaumann, 2020-08-19:
### Explanation
That one is simple. Every board definition can contain a definition of all the tools that are needed to compile the software for the specific board and hardware. If this is used then it nails down e.g. the version of the compiler used.
Digistump has done this and since the package definition is quite old the compiler used in that definition is quite old as well. The code that I've written uses newer compiler functionality and does not work with very old compiler versions.
### Fix
Depending on the system on which you run the IDE and on your knowledge of the underlying system different ways are possible to solve this. I would suggest one of the following:
1. If your system allows symbolic links and you know that you don't need the old compiler anymore you can substitute the old compiler directory with a link to the newest compiler version.
These directories can be found in `<Arduino Directory>/packages/arduino/tools/avr-gcc`. You can determine the Arduino Directory by looking for the location of the preferences.txt file in the Preference Pane.
2. Edit the package definition. This is named `package_digistump_index.json` and can be found in the Arduino Directory as well.
In the file look for the definition of the "Tool Dependencies" in "Digistump AVR Boards" or simply search for the string "4.8.1-arduino5". Change this to the name of the directory for your newest compiler, in my case "7.3.0-atmel3.6.1-arduino7".

These modifications should be done when the Arduino IDE is not running.

When you restart the Arduino IDE it should use the newest compiler and the next problem will be that the EEPROM-library cannot be found. Simply copy the library from the ATTinyCore definition to the Digistump definition and you should be good to go. The respective directories on my system are
`<Arduino Directory>/packages/ATTinyCore/hardware/avr/1.3.3/libraries`
 and `<Arduino Directory>/packages/digistump/hardware/avr/1.6.7/libraries`.

deHarro, 2020-08-19:
Wow, thanks a lot!
This did the trick.

I normally use Visual Studio with the VisualMicro extension to code for my Arduinos and such, but sometimes I fall back to the IDE.
Will this change be transparent for all other projects I have with Digisparks, or have I to revert to the original setting, when I work on other projects?

Can you please explain, why the deamon code is nailed to 8 MHz? (The Digispark has to run at 16,5 MHz to get the USB connection done (only at program time)).

Thanks!

Harald

jbaumann, 2020-08-20:
My brilliant idea in using the ArduinoIDE was to use something that poses a minimal problem even for beginners. I normally use IDEs that offer a minimum of user support (e.g., Visual Code, Eclipse, Intellij Idea...).

The change I suggested is a general change for all projects with Digisparks. If you are familiar with JSON structures, then simply copy the whole list entry (i.e., the object with the name "Digistump AVR Boards"), give it a new name e.g., "Digistump AVR Boards New Compiler" and change the dependency definition there.

This introduces a new menu with the given name and leaves the rest untouched.

Regarding the clock frequency: The datasheet says on p.163 in figure 21-1 and 21-2, and on  p. 172 in figure 22-2 that 2.7V and 10 MHz go hand in hand. If we want to guarantee that the processor runs without problems even at 3V or below, we cannot go above 8MHz. With 16MHz we would need a stable 4V, and this is simply not sensible with a LI-Ion battery that goes below 3V without damage.

deHarro, 2020-08-21:
That, too, was the idea behind my approach to use a Digispark as controller. If it had beeen feasible to program the controller without a dedicated programming device, that would have made it even easier for others to build your solution.
But the facts you write regarding the clock frequency kills that approch.

I have to confess, I rigged my test board wrong and didn't catch the drawbacks...
I used the 5V output of the HAT to power the Digispark, what runs perfectly well until the HAT is switched off :-)
The next step, switching the HAT back ON after the battery is recharged, wasn't on my mind til now.

Ok, so I have to revert to a standard AtTiny85, or at least remove some superflous parts on the Spark, which prevent it from normal operation, anyways. I.e. the zener diodes on two IOs for the USB stuff interfere with standard IO stuff.

Tonight I had another idea, but before starting to implement it, I better ask...
Why do you use an analog input to get the battery voltage, having I2C and the corresponding chip on the HAT on hands?
Let me guess... if the HAT is switched off, the chip doesn't answer I2C requests. True?

Thanks!
Harald

closed 2020-08-23 by jbaumann