jbaumann/attiny_daemon #9: Q: Replace battery with Super-capacitors?

ankostis, 2021-01-18:
@jbaumann  have you considere altering the design and avoid dependence on another charger circuit,
and rely instead on super-capacotors?
(something along the lines of https://juice4halt.com/)

jbaumann, 2021-01-19:
The ATTiny_Daemon is always depending of some kind of external buffer circuit that provides power for the RPi to shut down safely. This can be a LiIon or LiFePO4 battery with its charging circuit or a super capacitor circuit like the one you mentioned.  It can control this external circuit if possible and shut down and restart the RPi according to the state of that circuit.

One of the newest configuration options is that the ATTiny_Daemon can watch the external voltage and tell the RPi to shut down as soon as that voltage sinks below 4.7V. This would be the configuration option that I would use for that use case.

ankostis, 2021-01-19:
Thank you, you may closer it now.
If anyone has more resources (e.g. schematics), can add them here.

closed 2021-01-19 by jbaumann