jbaumann/attiny_daemon #10: Attiny85 correct board in ArduinoIDE

adrianmihalko, 2021-11-17:
Dear Joachim,

Which is the correct board version for us? In the Wiki you mention only Attiny 25/45/85, but there are multiple options. The most obvious is Attiny 25/45/85 but it is No bootloader version and we upload bootloader in the next step. I am a little bit confused here.

[image]

jbaumann, 2021-11-17:
Hi Adrian,

it is the version without the bootloader, because we use the whole available space for the firmware.

Cheers, Joachim

adrianmihalko, 2021-11-17:
Thank you very much. :)

adrianmihalko, 2021-11-17:
Let me ask you one more question. I put everything together and tested based on the guide.

I changed config:

```
button function = shutdown
ups configuration = 0x0
primed = true
```

ups configuration is 0x0 because I have Geekworm 1.0 board. Primed set to true after I checked I am getting data with ./showStateAndBattery.py. After that I restarted to make changes active.

I started by unplugging the power source, after 4 hour finally battery was empty enough to see if Attiny Daemon is working or not:

https://www.youtube.com/watch?v=QkljELmsSIg

As you can see for some it tries to boot at certain intervals. This is how it supposed to work? Isn't this bad for SD card? I thought it would be completely stay off after empty battery shutdown and reamins in this state until Attiny sense voltage.

After connecting to power source:

```
$ ./showStateAndBattery.py
INFO:root:Current state is 0x20: SHUTDOWN_STATE
INFO:root:Current should_shutdown value is 0x0
INFO:root:Current battery voltage is 3.839V.
INFO:root:Current external voltage is 5.566V.
INFO:root:Current warn voltage is 3.4V.
INFO:root:Current ups shutdown voltage is 3.2V.
INFO:root:Current restart voltage is 3.9V.
pi@octopi:/opt/attiny_daemon $ ./showStateAndBattery.py
INFO:root:Current state is 0x20: SHUTDOWN_STATE
INFO:root:Current should_shutdown value is 0x0
INFO:root:Current battery voltage is 3.881V.
INFO:root:Current external voltage is 5.594V.
INFO:root:Current warn voltage is 3.4V.
INFO:root:Current ups shutdown voltage is 3.2V.
INFO:root:Current restart voltage is 3.9V.
pi@octopi:/opt/attiny_daemon $ ./showStateAndBattery.py
INFO:root:Current state is 0x0: RUNNING_STATE
INFO:root:Current should_shutdown value is 0x0
INFO:root:Current battery voltage is 4.078V.
INFO:root:Current external voltage is 5.612V.
INFO:root:Current warn voltage is 3.4V.
INFO:root:Current ups shutdown voltage is 3.2V.
INFO:root:Current restart voltage is 3.9V.
```

jbaumann, 2021-11-17:
First of all, this behaviour is very bad for the SD card (or any storage medium). You should remove the orange cable (which is used to turn the UPS off and on) and then make further tests.

First observation: Controlling the UPS works. 

Second observation: I assume that the output has been done after you restarted the RPi, while the UPS was still charging the battery. This shows, that the state machine works as intended.

The video however, shows that the ATTiny tries to restart the RPi after a very short time. This could come from a timeout that is way too short. Could you post your full config file? After having run the daemon the config file should contain all relevant data.

And what is the output of the running attiny_daemon service?

I think you are 90% there and it might be only a little thing, but be careful not to stress your storage too much.

jbaumann, 2021-11-17:
The undervoltage is the result of the input voltage dropping below 4.65V. Since the UPS provides the input voltage, it could be a problem of the UPS trying to charge the battery at the same time as trying to feed the power to the RPi. The first UPS has the problem that it can provide only a limited amperage and thus might not be the right choice for a RPi3. If you are using the RPi without any USB devices you might get away with it, since the RPi itself uses 3.3V for nearly everything. What kind of power source are you using?

adrianmihalko, 2021-11-17:
> attiny_daemon

I haven't checked yet, but there are some interesting errors:

```
pi@octopi:/opt/attiny_daemon $ journalctl -r -u attiny_daemon
-- Logs begin at Wed 2021-11-17 18:17:01 CET, end at Wed 2021-11-17 20:15:21 CET. --
Nov 17 20:15:21 octopi attiny_daemon.py[345]: Sleeping for 20 seconds.
Nov 17 20:15:00 octopi attiny_daemon.py[345]: Sleeping for 20 seconds.
Nov 17 20:14:39 octopi attiny_daemon.py[345]: Sleeping for 20 seconds.
Nov 17 20:14:17 octopi attiny_daemon.py[345]: Sleeping for 20 seconds.
Nov 17 20:13:56 octopi attiny_daemon.py[345]: Sleeping for 20 seconds.
Nov 17 20:13:35 octopi attiny_daemon.py[345]: Sleeping for 20 seconds.
Nov 17 20:13:14 octopi attiny_daemon.py[345]: Sleeping for 20 seconds.
Nov 17 20:12:53 octopi attiny_daemon.py[345]: Sleeping for 20 seconds.
Nov 17 20:12:53 octopi sudo[948]: pam_unix(sudo:auth): auth could not identify password for [pi]
Nov 17 20:12:53 octopi sudo[948]: pam_unix(sudo:auth): conversation failed
Nov 17 20:12:53 octopi attiny_daemon.py[345]: sudo: no tty present and no askpass program specified
Nov 17 20:12:53 octopi attiny_daemon.py[345]: shutting down now...
Nov 17 20:12:47 octopi attiny_daemon.py[345]: Unknown shutdown_level 130. Shutting down.
Nov 17 20:12:26 octopi attiny_daemon.py[345]: Sleeping for 20 seconds.
Nov 17 20:12:26 octopi sudo[934]: pam_unix(sudo:auth): auth could not identify password for [pi]
Nov 17 20:12:26 octopi sudo[934]: pam_unix(sudo:auth): conversation failed
Nov 17 20:12:26 octopi attiny_daemon.py[345]: sudo: no tty present and no askpass program specified
Nov 17 20:12:26 octopi attiny_daemon.py[345]: shutting down now...
Nov 17 20:12:26 octopi attiny_daemon.py[345]: Couldn't set 8 bit register after 9 retries.
Nov 17 20:12:06 octopi attiny_daemon.py[345]: Unknown shutdown_level 130. Shutting down.
Nov 17 20:11:45 octopi attiny_daemon.py[345]: Sleeping for 20 seconds.
Nov 17 20:11:45 octopi sudo[915]: pam_unix(sudo:auth): auth could not identify password for [pi]
Nov 17 20:11:45 octopi sudo[915]: pam_unix(sudo:auth): conversation failed
Nov 17 20:11:45 octopi attiny_daemon.py[345]: sudo: no tty present and no askpass program specified
Nov 17 20:11:45 octopi attiny_daemon.py[345]: shutting down now...
Nov 17 20:11:45 octopi attiny_daemon.py[345]: Couldn't set 8 bit register after 9 retries.
Nov 17 20:11:25 octopi attiny_daemon.py[345]: Unknown shutdown_level 130. Shutting down.
Nov 17 20:11:04 octopi attiny_daemon.py[345]: Sleeping for 20 seconds.
Nov 17 20:11:04 octopi attiny_daemon.py[345]: sudo: no tty present and no askpass program specified
Nov 17 20:11:04 octopi attiny_daemon.py[345]: shutting down now...
Nov 17 20:11:04 octopi attiny_daemon.py[345]: Couldn't set 8 bit register after 9 retries.
Nov 17 18:17:31 octopi attiny_daemon.py[345]: Battery is at warn level. Shutting down.
Nov 17 18:17:30 octopi attiny_daemon.py[345]: Merging completed
Nov 17 18:17:25 octopi attiny_daemon.py[345]: Couldn't read 16 bit register 0x14 correctly.
Nov 17 18:17:10 octopi attiny_daemon.py[345]: Merge Values and save if necessary
Nov 17 18:17:10 octopi attiny_daemon.py[345]: ATTiny firmware version 2.13.14
```

With an empty battery after 2 minute of boot:

```
pi@octopi:/opt/attiny_daemon $ date
Wed Nov 17 20:12:35 CET 2021
pi@octopi:/opt/attiny_daemon $ ./showStateAndBattery.py
INFO:root:Current state is 0x20: SHUTDOWN_STATE
INFO:root:Current should_shutdown value is 0x82
INFO:root:Current battery voltage is 3.81V.
INFO:root:Current external voltage is 5.57V.
INFO:root:Current warn voltage is 3.4V.
INFO:root:Current ups shutdown voltage is 3.2V.
INFO:root:Current restart voltage is 3.9V.
pi@octopi:/opt/attiny_daemon $ date
Wed Nov 17 20:13:57 CET 2021
pi@octopi:/opt/attiny_daemon $ ./showStateAndBattery.py
INFO:root:Current state is 0x0: RUNNING_STATE
INFO:root:Current should_shutdown value is 0x2
INFO:root:Current battery voltage is 3.957V.
INFO:root:Current external voltage is 5.572V.
INFO:root:Current warn voltage is 3.4V.
INFO:root:Current ups shutdown voltage is 3.2V.
INFO:root:Current restart voltage is 3.9V.
pi@octopi:/opt/attiny_daemon $ date
Wed Nov 17 20:18:12 CET 2021
[... 130 lines cut ...]
```

[attinydaemon]
button function = shutdown
ups configuration = 0x0
primed = true
external voltage constant = 700
external voltage coefficient = 2000
sleeptime = 20
ups shutdown voltage = 3200
timeout = 120
pulse length = 4000
pulse length on = 500
pulse length off = 4000
temperature coefficient = 1000
i2c address = 0x37
battery voltage constant = 0
restart voltage = 3900
warn voltage = 3400
temperature constant = -270
battery voltage coefficient = 1000
force shutdown = 0
switch recovery delay = 1000
loglevel = DEBUG
led off mode = 0

adrianmihalko, 2021-11-17:
Progress:

Removed sudo from attiny_daemon.py from there:

```
_shutdown_cmd = "systemctl poweroff"  # sudo allows us to start as user 'pi'
_reboot_cmd  = "systemctl reboot"     # sudo allows us to start as user 'pi'
```

in attiny_daemon.service I changed User=pi to User=root.

Now shutdown works perfectly:

```
pi@octopi:/opt/attiny_daemon $ journalctl -f -u attiny_daemon
-- Logs begin at Wed 2021-11-17 22:03:16 CET. --
Nov 17 22:03:22 octopi systemd[1]: Started ATTiny UPS Daemon.
Nov 17 22:03:23 octopi attiny_daemon.py[347]: config variables are set
Nov 17 22:03:23 octopi attiny_daemon.py[347]: ATTiny Daemon version 2.13.1
Nov 17 22:03:25 octopi attiny_daemon.py[347]: ATTiny firmware version 2.13.14
Nov 17 22:03:25 octopi attiny_daemon.py[347]: Merge Values and save if necessary
Nov 17 22:03:44 octopi attiny_daemon.py[347]: Merging completed
Nov 17 22:03:45 octopi attiny_daemon.py[347]: Sleeping for 20 seconds.
Nov 17 22:04:19 octopi attiny_daemon.py[347]: Sleeping for 20 seconds.
Nov 17 22:04:40 octopi attiny_daemon.py[347]: Sleeping for 20 seconds.
Nov 17 22:05:01 octopi attiny_daemon.py[347]: Sleeping for 20 seconds.
Nov 17 22:05:22 octopi attiny_daemon.py[347]: Sleeping for 20 seconds.
Nov 17 22:05:43 octopi attiny_daemon.py[347]: Sleeping for 20 seconds.
Nov 17 22:06:04 octopi attiny_daemon.py[347]: Sleeping for 20 seconds.
Nov 17 22:06:25 octopi attiny_daemon.py[347]: Sleeping for 20 seconds.
Nov 17 22:06:46 octopi attiny_daemon.py[347]: Unknown shutdown_level 130. Shutting down.
Connection to octopi.local closed by remote host.
Connection to octopi.local closed.
```

I still don't know what is the unknown shutdown_level, but at least it is working now.

At 3.94V (measured with DMM) it started automatically.

can you tell me if this is how supposed to look like when it is off:

[image]

- UPS green led ON
- Pi power red led ON
- Attiny communication led blinking every 1s
- Charge indicator led is ON

Undervoltage:

[Wed Nov 17 22:14:26 2021] Under-voltage detected! (0x00050005)
[Wed Nov 17 22:14:43 2021] Voltage normalised (0x00000000)
[Wed Nov 17 22:14:51 2021] Under-voltage detected! (0x00050005)
[Wed Nov 17 22:14:55 2021] Voltage normalised (0x00000000)
[Wed Nov 17 22:21:03 2021] Under-voltage detected! (0x00050005)
[Wed Nov 17 22:21:08 2021] Voltage normalised (0x00000000)

Nothing is connected at this time. Charger is an Anker 2 or 2.5Ah? I think I get this warning when it is charges the battery. Battery charging + powering Pi at the same time is maybe too much for this charger.

jbaumann, 2021-11-18:
Which version of Raspbian are you using? I have never had the problem of having to provide a password for sudo from the user pi. I would have to look into that. Running a service as user root is generally a "bad idea"(TM) but in this case, until we have fixed this small problem, it is the right way to go.

The shutdown level is 0x80 + 0x2, which is "battery voltage low" and "RPI initiated shutdown". This should never be read together, because the RPi normally shuts the system down and then this value gets reset. In your case shutting down fails and thus this value is read, but correctly identified as a value indicating a necessary shutdown. I simply have to add a message that states the correct reason (never thought of this...).

The read error directly after the shutdown message looks a bit strange, I'll look into it. But this isn't something to worry about, because the communication is reset on both sides after half the timeout and thus re-established.

The state of the system when off is correct with your configuration. You can set "force shutdown = 1" to tell the ATTiny that it should turn off the UPS as soon as the shutdown voltage has been reached.

jbaumann, 2021-11-19:
Could you check whether you have a file /etc/sudoers.d/010_pi-nopasswd ?

Here on my system:
```
pi@kodi:/etc $ cd sudoers.d/
pi@kodi:/etc/sudoers.d $ dir
010_at-export  010_pi-nopasswd	010_proxy  pihole  README
pi@kodi:/etc/sudoers.d $ sudo cat 010_pi-nopasswd
pi ALL=(ALL) NOPASSWD: ALL
pi@kodi:/etc/sudoers.d $
```
This is the default configuration on Raspbian.

adrianmihalko, 2021-11-19:
Mine seem to be commented out:

```
pi@octopi:/etc/sudoers.d $ ls
010_at-export  010_pi-nopasswd  010_proxy  README  octoprint-service  octoprint-shutdown
pi@octopi:/etc/sudoers.d $ sudo cat 010_pi-nopasswd
[sudo] password for pi:
#pi ALL=(ALL) NOPASSWD: ALL
```

I remembered that I have not a raw Raspbian, but instead Octopi (which is based on Raspbian). So I started to search and what I found? A ticket opened by me https://github.com/guysoft/OctoPi/issues/549 about same issue.

Let's close this issue then. Otherwise now everything works as it should:

```
Nov 19 16:21:19 octopi attiny_daemon.py[354]: Sleeping for 20 seconds.
Nov 19 16:20:58 octopi attiny_daemon.py[354]: Sleeping for 20 seconds.
Nov 19 16:20:57 octopi attiny_daemon.py[354]: Couldn't read register 0x23 correctly: 0x7f
Nov 19 16:20:36 octopi attiny_daemon.py[354]: Sleeping for 20 seconds.
Nov 19 16:20:15 octopi attiny_daemon.py[354]: Sleeping for 20 seconds.
Nov 19 16:19:54 octopi attiny_daemon.py[354]: Sleeping for 20 seconds.
Nov 19 16:19:33 octopi attiny_daemon.py[354]: Sleeping for 20 seconds.
Nov 19 16:19:12 octopi attiny_daemon.py[354]: Sleeping for 20 seconds
```

jbaumann, 2021-11-19:
Fantastic. Good to hear.

closed 2021-11-19 by jbaumann

adrianmihalko, 2021-11-24:
One more question, button function is set to shutdown, but it doesn't work for some reason. If I press I see green activity LED is blinking, but the system does nothing. Nothing is in the logs.

jbaumann, 2021-11-25:
I assume that the daemon is running as a service. Could you use the script ‘showStateAndBattery.py’ to print out the state and shutdown value before and after a button press? Normally the pressed button is acknowledged on the ATTiny side with the LED blinking, and this information is transferred to the RPi with the next communication between daemon and ATTiny. This directly leads to a log entry that the button has been pressed and that the configured action will be executed. So, even if you have no action configured, the log still shows that the button has been pressed.

Cheers, Joachim

adrianmihalko, 2022-09-09:
Hi @jbaumann. I am building another UPS. Which UPS board are you currently recommend? I see you mention Geekworm's UPS3 on index page, but there is no installation guide for that?

jbaumann, 2022-09-10:
Hi @adrianmihalko,

sorry, I never got around to add pictures for the UPS3. It is so simple that I wouldn’t call it a modification. You need the battery power, which you get from the two battery power pads (+ and -) adjacent to the battery connector. Then you need the connection to the switch, which is exposed on an unpopulated header named “sw”. And that’s it. In fact, I’m still using the UPS3 in some of my installations.

Regarding the GeekPi UPS: I never bought a newer version of this UPS, so I cannot say anything new.

I’ve mainly worked on the SuperPower project, where we created a UPS from scratch. Sadly, even though the firmware has been finished, the hardware had some problems, and the second revision has never been finished.

But @Burgduino is currently developing a UPS that will have everything needed and is programmed by me. So, depending on the urgency of your new project you could wait for that…

adrianmihalko, 2022-09-14:
> Hi @adrianmihalko,
>
> sorry, I never got around to add pictures for the UPS3. It is so simple that I wouldn’t call it a modification. You need the battery power, which you get from the two battery [...]
Thanks for the detailed info.

> 
> But @Burgduino is currently developing a UPS that will have everything needed and is programmed by me. So, depending on the urgency of your new project you could wait for that…

I am sad because of the SuperPower project, it was very promising.

Where I can read more about @Burgduino's project? It is public?

jbaumann, 2022-09-14:
Hi @adrianmihalko, here is the link to his Discord server where he discusses the current status for many of his projects: https://discord.gg/bU5ph8EX

He calls his UPS the "Pi Vault". And he sells his stuff mainly through Lectronz and Tindie.