jbaumann/attiny_daemon #11: Strange behavior

suomi35, 2024-10-07:
I've built the circuit, flashed an ATTiny85, and connected everything (Raspberry Pi Zero 2 W + Geekworm 1.0).

I have a couple of issues:

1. I am able to power the system on via the button, but the shutdown behavior is not quite right. When I press the button to initiate shutdown the rPi does shutdown, but then gets powered back up after a period of time (a minute or so).
Is that the expected behavior? (I was expecting the rPi to remain powered off until I press the button again.)

2. I also observe that when I am running on battery alone, and the battery goes low, the rPi is shutdown correctly, but the ATTiny LED keeps flashing at once-per-second, 'never' tuning off. If I then plug USB power back into the geekworm, nothing changes; the rPi remains off and the LED continues to blink once-per-second. Pressing the ATTiny button has no effect, and I have to unplug both battery and USB from the geekworm in order to get the rPi to boot.

Here is my config file:
```
[attinydaemon]
button function = shutdown
ups configuration = 0x0
primed = 1
external voltage constant = 700
external voltage coefficient = 2000
sleeptime = 20
ups shutdown voltage = 3200
timeout = 120
pulse length = 200
pulse length on = 500
pulse length off = 4000
temperature coefficient = 1000
i2c address = 0x37
battery voltage constant = 0
restart voltage = 3900
warn voltage = 3400
temperature constant = -270
battery voltage coefficient = 1000
force shutdown = 0
switch recovery delay = 200
loglevel = ERROR
led off mode = 0
```

```
suomi35@test-pi:/opt/attiny_daemon $ sudo systemctl status attiny_daemon
● attiny_daemon.service - ATTiny UPS Daemon
     Loaded: loaded (/etc/systemd/system/attiny_daemon.service; enabled; vendor preset: enabled)
     Active: active (running) since Mon 2024-10-07 14:55:48 PDT; 21min ago
   Main PID: 340 (attiny_daemon.p)
      Tasks: 1 (limit: 191)
        CPU: 512ms
     CGroup: /system.slice/attiny_daemon.service
             └─340 /usr/bin/python3 -u /opt/attiny_daemon/attiny_daemon.py

Oct 07 14:55:48 test-pi systemd[1]: Started ATTiny UPS Daemon.
```

jbaumann, 2024-10-08:
When you set `primed` in the config file, the ATTiny tries to keep the RPi running as long as the battery voltage is high enough. So to have the button act as a On/Off button you first have to turn `primed` off.

Your second issue suggests that the voltage measurements might be off. You can check that using the script `ShowStateAndBattery.py` or `useATTiny.py` (which is more of a test program). You have to adjust the voltage calculation so that it is in the same ballpark as the actual voltage. You can do that with the battery voltage constant and coefficient(see Wiki).

suomi35, 2024-10-08:
@jbaumann thank you for the lightning fast response! 🤘
Setting `primed = 0` fixed the button issue immediately.

I will study the constant/coefficient section of the wiki and see where I need to move the voltage calculation.
(Also, I am currently monitoring the Vcc line from the Geekworm through the voltage divider to pin 2 on the ATTiny...maybe that is silly, given that it should always be 5v or nothing.)

closed 2024-10-09 by suomi35